rclex/rclex #6: macOS support

takasehideki, 2020-02-26:
Should we support macOS (Mojave or Catalina) platform?

I believe Alchemists (Elixir programmers) want to use macOS instead of Ubuntu. macOS is also useful to develop web applications.
ROS 2 supports macOS as Tier 1 with binary packages. Although [REP 2000 for Dashing](https://www.ros.org/reps/rep-2000.html#dashing-diademata-may-2019-may-2021) only targets on Sierra, Mojave is also fine by my experience (I have never checked it on Catalina).

zacky1972, 2020-02-26:
It depends on how much code differences you have to modify to port it.
In the other word, it depends on how much your NIFs depends on platform specific notations.
If it will be too much, we consider to use it in a Docker container.

takasehideki, 2020-02-26:
@zacky1972 Thanks to your opinion!
I think NIFs in Rclex do not depend on kernel so much because ROS 2 is only middleware for robotics communication. Differences about the location and specification of shared library files may let some confusions, but it would not be problem for us. 
One of the expected issues we afraid is that increasing supported platform will introduce the cost to be tested precisely. However, we should pay an effort for macOS support if there are many alchemist would like to use this!!

In other words that we ask this to you;
Do you want to use the Rclex package and ROS2 on macOS environments?

zacky1972, 2020-02-26:
Does bindings for other languages support themselves on macOS environments?
I guess you should follow it.

takasehideki, 2020-02-26:
C++ and Python, which are officially supported languages by Open Robotics, surely support to be operated on multiple platform (macOS and Windows). We don't wanna consider Windows yet,,,

takasehideki, 2021-07-14:
After too long time, the situation has changed drastically. ROS community has decided to lower the priority of support for macOS because it is difficult to keep up with the versions of many dependent libraries published by homebrew.
https://www.ros.org/reps/rep-2000.html#galactic-geochelone-may-2021-november-2022

We decided to give up this issue and close it. Please let us know if you have strong suggestions about this.

closed 2021-07-14 by takasehideki