rclex/rclex #111: Refactoring and restructuring suggestions

s-hosoai, 2022-05-20:
使う際に気になったところをざっとピックアップしました。例示していないモジュールも同様に再整理します。
多少API構成も変わってしまいますが、どうでしょう？
[image]

takasehideki, 2022-05-20:
はい，とても良いと思います．当初からツギハギでここまで来ているので,,, 要素技術が出揃ってきたこの機会で大幅にリファクタするのに大賛成です．
ちな困ったときは Rclpy の仕様とか実装を見ています．
https://docs.ros2.org/foxy/api/rclpy/index.html

1. ここがいちばん悩ましいですね．API的には Node や Timer を見せる必要があり ResourceServer はライブラリ内部の機構としてあるのが望ましい（ユーザからは見えなくてよい）と思っています．
2. 上記と同様ですね．
3. これはおっしゃるとおりで．dead code も大量にあると思います．
4. ここは現状の仕様がどうなってるかの多少は分かる資料があるはずなので，あとで探して共有します．
5. これは大賛成です．Rclpy でも複数形を使っているみたいです． https://docs.ros2.org/foxy/api/rclpy/api/node.html#rclpy.node.Node.declare_parameter

takasehideki, 2022-05-27:
fixed by #118

closed 2022-05-27 by takasehideki