rclex/rclex #138: Unlock the version of git_hooks after its issue is fixed

takasehideki, 2022-07-09:
This issue is just a memo for future work.

Since there seems to be an issue in the latest version of elixir_git_hooks, we locked its version to 0.6.5 https://github.com/rclex/rclex/commit/aac946f358ddf5307bf2f9f5d8a90cb93113d185

So we will unlock the version of git_hooks when https://github.com/qgadrian/elixir_git_hooks/issues/123 is fixed, by reopeing https://github.com/rclex/rclex/pull/137

qgadrian, 2022-11-18:
@takasehideki there is a pre-release version `0.8.0-pre0` that hopes to get the dependencies issues fixed. 

Could you please take a look at it and confirm if it's working for you? I want to ensure that it will work for every use case

takasehideki, 2022-12-03:
@qgadrian We are deeply grateful for your careful announcement to our repository. I have tried `0.8.0-pre0` in my local environment and have not experienced any issues so far. I will share any further issues as they arise.
I think it's still a pre-release now, and I will update your package in our repository when it becomes official. Thanks again for the great package you've created!

qgadrian, 2022-12-05:
Thanks for the info @takasehideki 

I'm checking one additional issue with another user, I will ping you back when I have a production-ready version to release 🙌

takasehideki, 2025-02-25:
@qgadrian sorry for my too late reaction,, 
To be honest, we no longer use `git_hooks` in this repository, but when I tried this in another repository, I confirmed that the latest version works fine!! Thank you very much for your wonderful contribution 🙇‍♂ 
https://github.com/rclex/rclex_docker/pull/19

takasehideki, 2025-02-25:
I found that `git_hooks` has been removed in the breaking change at v0.10.0.
https://github.com/rclex/rclex/compare/v0.9.3...v0.10.0#diff-dfa6f4ed74c90e5d4fda283d547d366586e690387289bcfd473e3fa5f9ace308

@pojiro What do you think about rebasing it, adding `git_hooks` again to this repository?

pojiro, 2025-02-25:
@takasehideki 

> What do you think about rebasing it, adding git_hooks again to this repository?

When we do next batch of development, I think it would be good to enable `git_hooks` again.

takasehideki, 2025-03-04:
Reapplied `git_hooks` into this repository at #366 !

PR #366 (2025-03-04): Reapply git_hooks (v0.8.0 that has been addressed about path issue)
relate
https://github.com/rclex/rclex/issues/138#issuecomment-2681814627

will close #138
closed 2025-03-04 by takasehideki