rclex/rclex #228: Bump to Iron Irwini

takasehideki, 2023-05-24:
Happy World Turtle Day! The new ROS 2 Release "Iron Irwimi" has come!
https://discourse.ros.org/t/ros-2-iron-irwini-released/31553

We need to check the operation with Iron installed on the host, and if it does not work, issues should be clarified and resolved.
Also, ROS 2 Foxy will soon be sunset. If it is difficult to manage the code to support both, we may eliminate the code for Foxy.

related: #114

takasehideki, 2023-09-12:
Currently, we have confirmed that Rclex with Iron could work fine with change #251, only on the natively installed environment for Ubuntu 22.04 and Iron.

However, due to the following issues, we decided that the support of Iron is experimental; we think it is not so problem because Iron is STS release until Nov 2024.
- `mix test` may be locked only in the docker environment
  - The automatic test should run after building app, but test cannot be operated before that (the operation may be blocked). More mysteriously, after canceling the test with Ctrl+C and repeating the execution several times, the test can be run normally. We guess the issue may be about LTTng, by observing `export LTTNG_UST_DEBUG=1`.
  - The same issue has occurred on GitHub Actions because it requires the docker environment. So we gave up on adding Iron as the CI target on GitHub Actions. 
- rclex_on_nerves could not work fine
  - Nerves firmware hangs after startup and reboots repeatedly.
  - We guess the issue may also be LTTng,,, Additional configuration or kernel modules may be required by customizing nerves_system_br. Or, we need to prepare a customized ROS 2 Iron Docker image that does not use LTTng.

Note: changes about tracetools in Iron Irwini
https://docs.ros.org/en/iron/Releases/Release-Iron-Irwini.html#tracetools

takasehideki, 2023-09-12:
Note: how to add LTTng lib in rclex_on_nerves, but this alone was not enough to handle.

```
  defp vendor_resources(arch, "iron") do
    dir_name = arch_dir_name(arch)

    [
      "/lib/#{dir_name}/libspdlog.so*",
      "/lib/#{dir_name}/libtinyxml2.so*",
      "/lib/#{dir_name}/libfmt.so*",
      # humble needs OpenSSL 3.x which Nerves doesn't have
      "/lib/#{dir_name}/libssl.so*",
      "/lib/#{dir_name}/libcrypto.so*",
      "/lib/#{dir_name}/liblttng-*.so*"
    ]
  end
```

takasehideki, 2023-10-03:
ping to #256: this PR added ROS_INCS path to build with Iron properly.

pojiro, 2023-10-04:
NOTE

In https://github.com/rclex/rclex/issues/228#issuecomment-1715288806

> mix test may be locked only in the docker environment

is not correct. When i develop 0.10.0,  I faced the block situation on Ubuntu22.04 (not Docker) by following command

```
mix clean && mix test
...
Compiling 12 files (.ex) # blocked here
```

This block appears only first time. After `Ctrl + C`,  next invokation of `mix clean && mix test` was successfull.

takasehideki, 2023-10-04:
@pojiro thank you for your suggestion, yours is correct.
I changed the name of this issue properly.