rclex/rclex #253: Trying to make it work for ROS iron and elixir 1.15.6-otp-26

edwinallenz, 2023-09-29:
I'm trying to run the package with the adlkfj versions.
However on a clean install after running `mix deps.compile` I get the following error

`mix deps.compile
==> rclex
cc -o /home/ed/projects/ros_elixir_ws/src/hello-elixir/src/talker/_build/dev/lib/rclex/obj/graph_nif.o -c -g -O2 -Wall -Wextra -Wno-unused-parameter -pedantic -fPIC -I./src -I/home/ed/.asdf/installs/erlang/26.1.1/usr/include -I/opt/ros/iron/include/rcl -I/opt/ros/iron/include/rcl_yaml_param_parser -I/opt/ros/iron/include/rcutils -I/opt/ros/iron/include/rmw -I/opt/ros/iron/include/rosidl_runtime_c -I/opt/ros/iron/include/rosidl_typesupport_interface -I/opt/ros/iron/include/rosidl_dynamic_typesupport -I/opt/ros/iron/include/actionlib_msgs -I/opt/ros/iron/include/action_msgs -I/opt/ros/iron/include/diagnostic_msgs -I/opt/ros/iron/include/geometry_msgs -I/opt/ros/iron/include/lifecycle_msgs -I/opt/ros/iron/include/map_msgs -I/opt/ros/iron/include/nav_msgs -I/opt/ros/iron/include/pcl_msgs -I/opt/ros/iron/include/pendulum_msgs -I/opt/ros/iron/include/rosgraph_msgs -I/opt/ros/iron/include/sensor_msgs -I/opt/ros/iron/include/service_msgs -I/opt/ros/iron/include/shape_msgs -I/opt/ros/iron/include/statistics_msgs -I/opt/ros/iron/include/std_msgs -I/opt/ros/iron/include/stereo_msgs -I/opt/ros/iron/include/tf2_geometry_msgs -I/opt/ros/iron/include/tf2_msgs -I/opt/ros/iron/include/tf2_sensor_msgs -I/opt/ros/iron/include/trajectory_msgs -I/opt/ros/iron/include/unique_identifier_msgs -I/opt/ros/iron/include/visualization_msgs src/graph_nif.c
In file included from /opt/ros/iron/include/rcl/rcl/rcl.h:77,
                 from src/graph_nif.c:11:
/opt/ros/iron/include/rcl/rcl/node.h:36:10: fatal error: type_description_interfaces/srv/get_type_description.h: No such file or directory
   36 | #include "type_description_interfaces/srv/get_type_description.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:78: /home/ed/projects/ros_elixir_ws/src/hello-elixir/src/talker/_build/dev/lib/rclex/obj/graph_nif.o] Error 1
could not compile dependency :rclex, "mix compile" failed. Errors may have been logged above. You can recompile this dependency with "mix deps.compile rclex --force", update it with "mix deps.update rclex" or clean it with "mix deps.clean rclex"
==> 
** (Mix) Could not compile with "make" (exit status: 2).
You need to have gcc and make installed. If you are using
Ubuntu or any other Debian-based system, install the packages
"build-essential". Also install "erlang-dev" package if not
included in your Erlang/OTP version. If you're on Fedora, run
"dnf group install 'Development Tools'".`

erlang-dev and build-essential is already installed.

Would like to know any guidance on how to get it working

takasehideki, 2023-10-03:
Hi @edwinallenz, nice to meet you, and thank you so much for trying out our library!!

When I tried to do `mix compile` in this library in my native environment (where I installed your versions directly, elixir 1.15.6-otp-26 / erlang 26.1.1 by asdf and ros-iron-desktopy by apt install), I confirmed that your issue has unfortunately reproduced,,, I will investigate this issue;(

BTW, why do you want to use these versions, especially for ROS 2 Iron?
We recommend using the following version, with ROS 2 Humble that is LTS as ROS 2 distribution.
https://github.com/rclex/rclex#the-environment-where-host-development-and-target-operation-are-the-same

Since Iron is an STS release, we do not want to support this with higher priority due to our limited development resources. However, please feel free to ask us if you have a strong request about the usage of our library with Iron.

takasehideki, 2023-10-03:
@edwinallenz Good news! I can fix your issue by modifying Makefile according to the below.

```
diff --git a/Makefile b/Makefile
index 21aedfa..0be9b7f 100644
--- a/Makefile
+++ b/Makefile
@@ -33,7 +33,7 @@ ifeq ($(ROS_DISTRO), humble)
 ROS_INCS    ?= rcl rcl_yaml_param_parser rcutils rmw rosidl_runtime_c rosidl_typesupport_interface
 ROS_CFLAGS  ?= $(addprefix -I$(ROS_DIR)/include/, $(ROS_INCS)) $(addprefix -I, $(wildcard $(ROS_DIR)/include/*_msgs))
 else ifeq ($(ROS_DISTRO), iron)
-ROS_INCS    ?= rcl rcl_yaml_param_parser rcutils rmw rosidl_runtime_c rosidl_typesupport_interface rosidl_dynamic_typesupport
+ROS_INCS    ?= rcl rcl_yaml_param_parser rcutils rmw rosidl_runtime_c rosidl_typesupport_interface rosidl_dynamic_typesupport type_description_interfaces builtin_interfaces
 ROS_CFLAGS  ?= $(addprefix -I$(ROS_DIR)/include/, $(ROS_INCS)) $(addprefix -I, $(wildcard $(ROS_DIR)/include/*_msgs))
 else
 ROS_CFLAGS  ?= -I$(ROS_DIR)/include
```

I'm not fully sure why it is, but I guess the change in Iron about "Type Description Distribution" may cause your issue.
https://docs.ros.org/en/rolling/Releases/Release-Iron-Irwini.html#type-description-distribution

I will create PR to fix this.

takasehideki, 2023-10-03:
@edwinallenz Could you confirm the operation works fine with #256?
Editing `mix.exs` is the simplest way to try it out.

```
      {:rclex, git: "https://github.com/rclex/rclex.git", branch: "fix_253"}
```

closed 2023-10-03 by takasehideki

edwinallenz, 2023-10-06:
@takasehideki thank you so much for the fast response and fix, it worked. I would never sort it out.

> BTW, why do you want to use these versions, especially for ROS 2 Iron?
Because is the latest version and even if it is not LTS when I start exploring any new library or service I always try to stick to the most recent release, given conceptual changes that are sometimes introduced and tried on latest versions. Yes, I read that the iron is not supported, but given that my goal at this point is to start learning and playing, I decided to give it a try.

I'm in the middle of building and launching an [AR4](https://www.anninrobotics.com/) robotic arm. I've been developing on Elixir for the last 5 years and been an Elixir fan, I was looking to find a good way to use some Elixir along with ros, given all the concepts from ros that Elixir already manages, from the comments of this [post](https://dev.to/jimsy/elixir-for-robots-544e), so trying to explore I found you're library.

BTW, I read that you will be presenting  Rclex on Nerves, on Ros conf, would be good if you share access to those materials, afterward.

best

edwinallenz, 2023-10-06:
worked well
[image]

takasehideki, 2023-10-06:
@edwinallenz 🎉 
It's always great fun for us to observe our product working on different grounds.

I'm so curious about your project. We have applied Rclex to ground wheel robots, but have not yet tried it on arm robots. You may want to use msg containing arrays (e.g., JointState) or Services, but these remain on our TODO list. And while Iron is certainly the latest, it has incorporated LTTng as a standard, which is becoming more difficult to deal with, especially in Nerves (we are aware of a solution, but have yet to implement it).
Anyway, please feel free to contact and/or request us whatever you want to realize. We want to help you according to your demand!!

> BTW, I read that you will be presenting Rclex on Nerves, on Ros conf, would be good if you share access to those materials, afterward.

Yes, obviously!!
Before that, I wanna share our past talks that have been given at past Code BEAM: the former in 2021 introduces this Rclex library, and the latter in 2022 further introduces support for Nerves (and extensions to wide-area distributed communications leveraging Zenoh). I hope you can catch up on our activity from them.
These past talks have been tended toward Elixir/Erlang developers, but at this ROSCon we intend to bring all these contents together and present them to ROS developers. As a major update to be added them, I will introduce a case study of a ground wheel robot, but the presentation time of ROSCon is limited to 10 minutes, very tight,,,;(

commit 51159a8e (2023-10-03): Merge pull request #256 from rclex/fix_253

add ROS_INCS path for using with Iron properly (fix #253)
commit f7936e65 (2023-10-03): add ROS_INCS path for using with Iron properly (fix #253)

See also https://github.com/rclex/rclex/issues/253

I'm not fully sure why it is, but I guess the change in Iron about "Type Description Distribution" may cause your issue.
https://docs.ros.org/en/rolling/Releases/Release-Iron-Irwini.html#type-description-distribution
PR #257 (2023-10-03): release v0.9.2
## v0.9.2 on 03 Oct 2023

**Full Changelog**: https://github.com/rclex/rclex/compare/v0.9.1...v0.9.2

* New features: none
* Code Improvements/Fixes:
  * add ROS_INCS path for using with Iron properly (fix #253) by @takasehideki in https://github.com/rclex/rclex/pull/256
* Bumps:
  * Bump credo from 1.7.0 to 1.7.1 by @dependabot in https://github.com/rclex/rclex/pull/254
* Known issues to be addressed in the near future:
  * Lock `git_hooks` to 0.6.5 due to its issue in https://github.com/rclex/rclex/issues/138
  * Release rcl nif resources when GerServer terminates in https://g [... cut]
commit 9e2760dc (2023-10-03): release v0.9.2

**Full Changelog**: https://github.com/rclex/rclex/compare/v0.9.1...v0.9.2

* New features: none
* Code Improvements/Fixes:
  * add ROS_INCS path for using with Iron properly (fix #253) by @takasehideki in https://github.com/rclex/rclex/pull/256
* Bumps:
  * Bump credo from 1.7.0 to 1.7.1 by @dependabot in https://github.com/rclex/rclex/pull/254
* Known issues to be addressed in the near future:
  * Lock `git_hooks` to 0.6.5 due to its issue in https://github.com/rclex/rclex/issues/138
  * Release rcl nif resources when GerServer terminates in https://github.com/rclex/rclex/iss [... cut]