rclex/rclex #281: About Qos impl and `enif_alloc`

pojiro, 2024-01-04:
@FelixPe

Thank you for your interest in Rclex.
Hi, I'm main developer of `0.10.0-dev` branch, sorry to not repling on #276. 

I impl Qos as map in elixir side and provide functions which create qos profile maps, like  sensor_data on #280.
If you have any idea or question about #280, please let us know on #280 (before merging) or here (after merging).

And I'm interested in your idea allocator.c/h on #276, so I'd be happy if you could make a PR if possible after merging of #280.
I would like to test and consider whether it is possible to incorporate it into our implementation, so I would like you to make a PR which add only those two files. Of course, it's fine if it's possible.

We are proceeding dev through trial and error, it tends to be difficult to merge big PRs. So if you create a PR in the future, I would be happy if you could share your ideas in an issue before creating one.

Thank you.

FelixPe, 2024-01-10:
Hi @pojiro, 
thank you for your message. I like the layered architecture of 0.10.0-dev, it is a pleasure to follow. 

I will update my PR for the allocator based on #280.

pojiro, 2024-01-11:
@FelixPe Thank you for reply.

> I will update my PR for the allocator based on #280.

I think you understand this, but just for safe, please create a PR for v0.10.0-dev where #280 was merged.

pojiro, 2024-01-11:
#280 was merged!

pojiro, 2024-01-15:
This issue will be fixed by #284 and #285.

pojiro, 2024-01-18:
#284 and #285 are merged to v0.10.0-dev. so I closed this issue.

@FelixPe Thank you so much for your contribution.

takasehideki, 2024-01-18:
@FelixPe @pojiro love this your contributions ❤️