rclex/rclex #344: Supporting sensor_msgs/Image

mlainez, 2024-10-02:
HI,

I'm trying to use the sensor_msgs/Image message type and am bumping into some issues. I am working on a library that would do something similar to what cv_bridge is doing for python, which is converting an Opencv image to a valid ros2 Image message.

I'm using Evision 0.2.9 with the latest rclex, on Elixir 1.17.2 on OTP27.

Using Evision, I can get an Evision.Mat that represents an image captured from my camera.
```elixir
camera = Evision.VideoCapture.videoCapture(0)
cv_picture = Evision.VideoCapture.read(camera)
```
I can see that the generated `Image.ex` file using `mix rclex.gen.msgs` is expecting the following typespec:

```elixir
@type t :: %__MODULE__{
  header: %Rclex.Pkgs.StdMsgs.Msg.Header{},
  height: non_neg_integer(),
  width: non_neg_integer(),
  encoding: String.t(),
  is_bigendian: byte(),
  step: non_neg_integer(),
  data: binary()
}
```
So I create my ros message with the following code:

```elixir
  defp create_ros_image_message(cv_picture) do
    stamp         = %Rclex.Pkgs.BuiltinInterfaces.Msg.Time{sec: :os.timestamp() |> elem(1), nanosec: :os.timestamp() |> elem(2)}
    frame_id      = "image"
    height        = cv_picture.shape |> elem(0)
    width         = cv_picture.shape |> elem(1)
    encoding      = "#{cv_picture.type |> elem(1)}#{cv_picture.type |> elem(0) |> to_string |> String.capitalize()}C#{cv_picture.channels}"
    is_bigendian  = <<0>>
    step          = (cv_picture.shape |> elem(2)) * (cv_picture.shape |> elem(1))
    data          = cv_picture |> Evision.Mat.to_binary()

    %Rclex.Pkgs.SensorMsgs.Msg.Image{
      header: %Rclex.Pkgs.StdMsgs.Msg.Header{stamp: stamp, frame_id: frame_id},
      height: height,
      width: width,
      encoding: encoding,
      is_bigendian: is_bigendian,
      step: step,
      data: data
    }
  end
```
Notice that I am using `Evision.Mat.to_binary` to get a binary representation of the image.

When I start my application, it fails with a `UnicodeConversionError` that comes from the following line in the generated `Image.ex`.

```elixir
{Rclex.Pkgs.StdMsgs.Msg.Header.to_tuple(header), height, width, ~c"#{encoding}", is_bigendian, step, data |> to_charlist()}
```

It fails because my binary cannot be converted to a charlist since it's not a valid `String`. From my understanding, we could replace `data |> to_charlist()` with `data |> :binary.bin_to_list()` and it should work for this message type as well as others.

However, when I make that change and recompile rclex, I am now getting `ArgumentError` coming from the Nif. See the example below:

```
** (ArgumentError) argument error
            (rclex 0.11.0) Rclex.Nif.sensor_msgs_msg_image_set!(#Reference<0.1081337418.891420676.67849>, {{{872037, 631914}, ~c"image"}, 480, 640, ~c"8UC3", <<0>>, 1920, [143, 139, 135, 143, 139, 135, 143, 138, 134, 145, 139, 135, 139, 139, 137, 141, 142, 139, 142, 143, 140, 142, 143, 140, 141, 146, 138, 141, 146, 138, 139, 146, 138, 140, 148, 140, 138, 148, 140, 139, 149, 141, 141, ...]})
            (rclex 0.11.0) lib/rclex/publisher.ex:67: Rclex.Publisher.handle_call/3
            (stdlib 6.0) gen_server.erl:2209: :gen_server.try_handle_call/4
            (stdlib 6.0) gen_server.erl:2238: :gen_server.handle_msg/6
            (stdlib 6.0) proc_lib.erl:329: :proc_lib.init_p_do_apply/3
```

Tinkering around in the generated Nif files, it seems to come from the Header, but I don't think I'm doing anything wrong here. It could however come from elsewhere. I'm ready to put some time into making this work and documenting how to do it but I don't really know where to start at this point. Does anyone have any pointers for me?

mlainez, 2024-10-02:
Hi, I just did a quick scan in past issues and PR and found out that `to_charlist()` on binary data was introduced in #325. If anyone that worked on this PR can help, it'd be greatly appreciated :+1:

pojiro, 2024-10-03:
Hi, @mlainez . I create a quick fix PR, #345 , for this issue. 

Can you confirm the branch it works for you? If not, can you write a test code which reproduce this issue?

pojiro, 2024-10-03:
Oh, I found the error.

I think you need to specify `is_bigendian  = <<0>>` as `is_bigendian  = 0`.`byte() is `0..255`.
ref. https://hexdocs.pm/elixir/1.12/typespecs.html#built-in-types

mlainez, 2024-10-03:
Hi @pojiro, thanks for the reply, I did notice the <<0>> error too and made the change when using FelixPe's fork. I'll test everything again today with your PR.

closed 2024-10-03 by mlainez

FelixPe, 2024-10-03:
@mlainez Thank you a lot for testing. Regarding the binaries in your ROS message, they are not converted on that dev branch, but directly copied. This needs further testing.

mlainez, 2024-10-04:
Hi @FelixPe, I did notice the binaries were not converted in your fork indeed, as well as the awesome work you put in supporting services and actions! Can't wait to see this in upstream, it would make rclex a pretty solid ros2 client.

mlainez, 2024-10-07:
Hello @FelixPe, I'm getting relatively poor performance when it comes to sending image messages, I tweaked the opencv parameters which improved the situation a bit, but it's still not great. I've been thinking about what you said:  "Regarding the binaries in your ROS message, they are not converted on that dev branch, but directly copied". Am I right to assume that the fact it is converting the binary to a list on the upstream could be a cause of the performance issues I'm noticing?

FelixPe, 2024-10-07:
The memory consumption should also improve by this change, as each list element requires an additional word. Regarding the performance, I added a small [benchmark test](https://github.com/FelixPe/rclex/blob/caf51b0442d89bebfa7511c25f0a3f37ac7bd92c/test/rclex/nif_benchmark_test.exs#L741).

The interfaces of the first try on action server/ client are heavily inspired by python, which might not be the best approach for the elixir world. Any feedback is more than welcome.

PR #345 (2024-10-03): Fix UnicodeConversionError caused by treating a uint8 array as Unicode
This might fix #344.

Don't merge untill #344 's disscussion is finished.

EDIT: #344 is finished as this PR works.